peterkvt80/vbit2 #1: Carousel subpage IDs

chris-y, 2016-12-08:
I've updated to vbit2, and have encountered an issue similar to https://github.com/peterkvt80/vbit-pi-stream/issues/2

In this version, all pages of a carousel are being seen by my TV as subpage 0001.
Usually my TV caches all subpages and I can select them at will, but this issue means they just refresh under subpage 1.

I'm undecided as to whether this is an improvement :) but it's certainly more consistent.

peterkvt80, 2016-12-09:
Subcodes were being taken from the root page of the carousel instead of the page being displayed. It should work now but I don't have a TV that can verify this.

ZXGuesser, 2016-12-09:
It seems to be outputting 0000 for every subpage now instead of 0001

peterkvt80, 2016-12-09:
The correct subcodes appear to be going into Packet::Header so unless that is messing things up I don't know what else could be wrong.

ZXGuesser, 2016-12-09:
hmm. When I output the subcode to stderr from Packet::Header it's always 0 for page 100

chris-y, 2016-12-09:
Looks mostly correct now, however I'm getting subpages numbered 1 less than they should be (ie. subpage 1 is numbered 0, etc)

peterkvt80, 2016-12-09:
Subcode numbers are replaced when loaded, the subcode numbers in the file are not used. I can fudge them so they start at 1 instead of 0. More annoying is when running on Windows the subcodes are still wrong and are only 0 or 1.

ZXGuesser, 2016-12-09:
Only the carousel subpages should start from 0001. Non-carousel pages should remain as subpage 0000.

ZXGuesser, 2016-12-10:
digging a bit deeper, on windows the subcodes are correct now - the problem is that only subpage 0001 is being output over and over rather than iterating through (teefax page 19A is convenient to watch as the four pages have their subcodes in the text)

peterkvt80, 2016-12-13:
In vbit2 the carousels and static pages are supposed to be handled separately. The carousel pages were getting into the static page sequence. This shouldn't have caused any visible problems except to slow down the static pages. I also fixed the subcode numbers to follow the spec. in Annex A1.

chris-y, 2017-02-06:
This all looks to be working fine to me!

closed 2017-02-06 by chris-y