peterkvt80/vbit2 #3: vbit2 gives up after about half an hour

chris-y, 2017-02-06:
Output from systemctl status shows no output since it stopped, yet still showing as running?!
A restart gets it working for another half hour.
```
pi@hestia:~/vbit2 $ sudo systemctl status vbit2
● vbit2.service - (null)
   Loaded: loaded (/etc/init.d/vbit2.sh)
   Active: active (running) since Sun 2017-02-05 22:17:10 UTC; 23h ago
  Process: 394 ExecStart=/etc/init.d/vbit2.sh start (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/vbit2.service
           ├─399 /bin/bash ./vb2
           ├─432 /home/pi/vbit2/vbit2 --dir /home/pi/teletext
           └─433 /home/pi/raspi-teletext-master/teletext -

Feb 06 18:17:43 hestia.cy2.uk vbit2.sh[394]: [Mag::GetPacket] Header thisSubcode=5
Feb 06 18:17:43 hestia.cy2.uk vbit2.sh[394]: [Carousel::nextCarousel] page 52101 cycle time=10
Feb 06 18:17:43 hestia.cy2.uk vbit2.sh[394]: [Mag::GetPacket] Header thisSubcode=1
Feb 06 18:17:44 hestia.cy2.uk vbit2.sh[394]: [Carousel::nextCarousel] page 52001 cycle time=8
Feb 06 18:17:44 hestia.cy2.uk vbit2.sh[394]: [Mag::GetPacket] Header thisSubcode=2
Feb 06 18:17:44 hestia.cy2.uk vbit2.sh[394]: [Carousel::nextCarousel] page 19a01 cycle time=4
Feb 06 18:17:44 hestia.cy2.uk vbit2.sh[394]: [Mag::GetPacket] Header thisSubcode=4
Feb 06 18:17:44 hestia.cy2.uk vbit2.sh[394]: mpp=19a subcode=4
Feb 06 18:17:44 hestia.cy2.uk vbit2.sh[394]: [Carousel::nextCarousel] page 80001 cycle time=8
Feb 06 18:17:44 hestia.cy2.uk vbit2.sh[394]: [Mag::GetPacket] Header thisSubcode=8
```

nat2001, 2017-02-27:
I noticed that it freezes every so often - don't remember how often - but I set up a crontab for evry half hour that restarts vbit2: sudo killall vbit2. The vb2 script file immediately restarts the script, so you'd hardly ever notice.

chris-y, 2017-03-10:
Killing vbit2 doesn't seem to restart it here, I've set cron to restart it every 30 mins instead until this is fixed.

ZXGuesser, 2017-03-10:
Are you invoking the vbit2 binary directly, or using the vb2 and vbit2.sh scripts generated by the install script? That should restart vbit2 any time the main process gets killed.

chris-y, 2017-03-10:
Using the scripts, but the fact it doesn't restart when it is killed isn't the problem here :)

ZXGuesser, 2017-03-10:
indeed, not the cause of it freezing, just strange that it isn't restarting! :)

ZXGuesser, 2017-03-10:
someone else told me they had a problem with the output dying after several seconds or minutes every time, and it turned out to be somehow caused by the software defined radio dongle plugged into the Pi, or its software.

nat2001, 2017-03-13:
What Raspberry Pi are you using? I'm using the model B (still) and discovered that overclocking it to 900mhz or more seems to improve the performance of vbit2 considerably.

ZXGuesser, 2017-03-13:
I have an original model B and a Pi 3, both of which ran vbit2 constantly throughout the Block Party event without freezing. That was basically the only thing running though, no desktop environment etc installed, just the usual handful of system daemons idling in the background.

nat2001, 2017-03-13:
I don't have desktop etc running, but it does display a testcard and play 'Ceefax music' as well as a web server...

ZXGuesser, 2017-03-13:
Heh, funny you should say that, mine were displaying a caption card image and the image I had tested them with before the event was of course test card F. I never got around to piping my collection of "pages from ceefax" MP3s through the system though. :)

peterkvt80, 2017-03-13:
If Vbit2 dies after half an hour it is likely to be a memory leak. Since I don't think that it happens on my system this could be due to a page. If you are running your own pages there may be a bug that you are exposing. I've seen that sort of thing where there are random extra PN commands which don't actually have any text in them.

chris-y, 2017-03-14:
I'm running Teefax with no modifications.  I'm also using an old Model B on an Ethernet cable with nothing else plugged in (and nothing else running - I think I started from the Jessie minimal image, however it did originally have vbit-pi-stream on it).  When I get my Pi0W and attach a composite output to it I'll start again and see if I get the same problem.

chris-y, 2017-04-19:
Incidentally since running this on a Pi0W I've only noticed this problem once - and I can't be sure the vbit2 service even started up properly in that instance.  It's definitely not consistently crashing after half an hour like it was previously (100% repeatable).

nat2001, 2018-10-31:
Funny enough, my system has started this since a re-install of the system (for a VBIT-Pi). Running 'top', it seems that pages start getting mixed up once vbit2's RAM usage exceeds 50%, and it eventually freezes up, requiring a 'sudo service vbit-py restart' to get it running again. I'll probably just re-install again tho :)

ZXGuesser, 2018-10-31:
Are you using a recent copy? If so I suggest rolling back to 9e1837ac67e898502e655f511ea2043415ed994b. Peter's recent subtitle stuff has broken things a bit.

nat2001, 2018-10-31:
OK, rolled back to the old version, it's running for now, but time will tell... There's definitely something up with memory on the latest version though, it's sitting at around 28% memory usage and P100 is regularly getting overwritten by P200. Of course, it could be something to do with my hardware...

trixster1979, 2019-03-09:
Hi, I’m seeing this issue too where vbit2 stops working after a few hours. This is on a Pi B with Stretch-Lite installed, nothing else is installed, Pi has an Ethernet connection.

I followed the instructions on this github page. Teefax seems to work great but after a few hours it just stops. I have to either kill and restart the service or reboot the Pi.

Can someone give me a steer on how to resolve this? I’d be happy with something which automates killing and restating the service, as mentioned above, but I’m not sure how to do that :)

nat2001, 2019-03-10:
For the automated method:
`crontab -e`
and enter
`0 * * * *  sudo killall vbit2`
This kills vbit every hour, but run.sh should automatically restart it fairly quickly. I don't think this method is really recommended, but it works!

ZXGuesser, 2020-01-12:
I suspect this was down to an intermittent issue in raspi-teletext https://github.com/ali1234/raspi-teletext/issues/20

Ensure you are running a fixed version of raspi-teletext (if you installed or [upgraded to](https://github.com/peterkvt80/vbit2/wiki/Updating) VBIT2 v2.3.0 you have a fixed version)

closed 2020-01-12 by ZXGuesser