peterkvt80/vbit2 #4: duplicated (or more) packets

ZXGuesser, 2017-03-04:
Sometimes an identical packet appears to be output twice or more.

ZXGuesser, 2017-08-19:
I think I've located the bug.
Service::run() calls pMag->GetPacket(pkt), and then tests to see if (pkt!=NULL). If I instead test if (pMag->GetPacket(pkt)!=NULL) it works properly. GetPacket isn't always setting the pointer to null when it returns null.
I feel like functions shouldn't return something that they were passed by reference, so I haven't just committed the fix.

ZXGuesser, 2017-08-19:
the issue is that we are doing this:
```
pointer* function(pointer *p){
  return null;
}

function(p);
if (p != NULL){
  do stuff
} else {
  never happens
}
```

The call is inside a loop, so when the function returns null, "do stuff" happens with the previous iterations 'p'.

I see two solutions (besides my bodgy fix above which leaves the function booby-trapped to allow the same mistake in future)
One is to remove the argument to the function, i.e: ```Packet* Mag::GetPacket()``` and do

```
pkt = GetPacket();
if (pkt != null){stuff}
```

The other is to do ```void Mag::GetPacket(Packet* p)``` and change all the instances of ```return null;``` into
```
p = null;
return;
```

Would the second be more efficient because it doesn't have to allocate memory for the packet each time?

peterkvt80, 2017-08-19:
I think I prefer the first option just because it is simpler.

ZXGuesser, 2017-08-19:
done :)

closed 2017-08-19 by ZXGuesser