peterkvt80/vbit2 #5: Newfor subtitles don't work

nat2001, 2017-04-08:
I can't get newfor subtitles to work with vbit2. I get: "java.net.ConnectException: Connection refused: connect".

peterkvt80, 2017-04-08:
Good point. Newfor subs are only in the original VBIT-Pi. vbit2 was a complete rewrite and that bit did not get implemented. I'll put it on my list of things to do.

nat2001, 2017-04-08:
OK

peterkvt80, 2017-08-27:
I now have a vbit2 branch called packet-stream that implements Newfor subtitles. Unfortunately I have made some major changes to the packet generating structure so it will be a big mess of conflicts. In the coming weeks I will try to do a merge without losing any of Alistair's contributions. I don't think it is going to be easy https://xkcd.com/1597/.

ZXGuesser, 2017-08-27:
Am I correct in thinking that the new system is going to replace Mag::GetPacket entirely? The merge shouldn't be too bad as it's only a few lines of stuff in mag and service that are presumably going to be redundant anyway?
It's not clear that my fix the other day actually solved the bug as I thought it had, but I held off fiddling with the code any further when I saw you were working in another branch to avoid making the merge a disaster.

peterkvt80, 2017-08-27:
The reorganisation is to make it easier to add Newfor subtitles and further down the line, non-page packets like 8/30.. GetPacket is now supplied by classes derived from PacketSource and PacketMag does most of the work that Mag::GetPacket and Service did. Instead of having Service run big pile of tests to decide what packet goes out when, each packet source just manages its own packets in its own way. This makes the Service thread very simple and encapsulates code better.

closed 2017-08-27 by nat2001