peterkvt80/vbit2 #6: Carousels

nat2001, 2017-08-01:
Since updating vbit2 yesterday (Monday), blank lines on carousel pages aren't transmitted, so pages have the remains of the previous page superimposed (if you know what I mean). Maybe this is normal and I'll have to change some of my own code?

ZXGuesser, 2017-08-01:
Sorry, this is the result of a change I made in among some other alterations the other day. Not transmitting blank rows is known as "row adaptive" transmission and is valid, from Annex D of the specification:
> If any decoder assumes the page is a basic text page, any missing rows in the range X/1 to X/23 should be displayed as if they contained 40 character spaces (code 2/0). Where required, packets X/24 and X/25 should be treated in the same way.

This causes a problem if the page files with carousels don't have the Erase bit set.
I'm working on adding user configuration options via config file at the moment, so perhaps a switch to disable row adaptive transmissions there would be a good idea.

peterkvt80, 2017-08-01:
It was a bit of a compromise. If you don't send blank rows then you can really optimise vbi usage. The BBC would only carousel the headline row and send the rest of the rows less frequently. But since the average user doesn't want to worry about rules for blank lines and doesn't care about wasting vbi lines I made it transmit blank rows. A hack to force it to send blank lines is to put a character like Alpha White in column 1. It still looks blank but the row is actually transmitted.

ZXGuesser, 2017-08-01:
Ok, I've written some code to use a config option for row adaptive mode, and set it to transmit blank rows by default. Just testing it now.

ZXGuesser, 2017-08-01:
ok, should be back to normal now, sorry about that :)

nat2001, 2017-08-01:
Sorry, didn't mean to cause any trouble. Thanks!

closed 2017-08-01 by nat2001