peterkvt80/vbit2 #8: Missing headers and duplicated packets

ZXGuesser, 2017-08-29:
The new implementation seems to be eating page headers, possibly other packets too. It's also duplicating packets again, I expect the same way as before where the return value of a function is ignored.

peterkvt80, 2017-08-29:
Spending 10 minutes I found and fixed a bug where the header would not be sent out if there was a row 27 packet in the page. Probably not the only bug but it looks a bit better now.

ZXGuesser, 2017-08-29:
That seems to have cured the missing headers, at least all the ones I had found.

ZXGuesser, 2017-08-29:
ok I've found where the duplicate packets are coming from. It's the `if (_page->IsCarousel() && _page->GetCarouselFlag()) // Don't let registered carousel pages into the main page sequence` condition in PacketMag::GetPacket.
Like before it is returning filler in several places but doesn't actually change the packet contents. I located the cause of these packets by going through one at a time adding `p->SetMRAG(8,25);` before each `return filler;` and looking at the output for filler packets with row data in appearing.

ZXGuesser, 2017-08-30:
I think I've got rid of the last of the duplicated packets now. We just need to be careful to keep on top of what we're returning from the various GetPacket functions as stuff is implemented to stop any creeping back in. I'll continue to keep an eye on what packets get output too :)

closed 2017-08-30 by ZXGuesser