peterkvt80/vbit2 #9: Vbit continually crashes

nat2001, 2017-08-31:
Running ./vb2 puts up the usual page adding messages, before ending with:
[Packet830::Packet830] constructor
[~TTXPage] none
[~TTXPage] none
[Service::worker]This is the worker process
[Command::run] Newfor subtitle listener started
[Command::run] Ready for a client to connect
[Service::run]Loop starts
[FileMonitor::run] Monitoring /home/pi/Pages
[Service::run] No packet available for this line
[Service::run] No packet available for this line
[Service::run] No packet available for this line
[Service::run] No packet available for this line
[Service::run] No packet available for this line
[Service::run] No packet available for this line
[Service::run] No packet available for this line
[Service::run] No packet available for this line
[FileMonitor::run]  Adding a new pageP363.TTI
[FileMonitor::run]  Adding a new pageMENU115.tti
terminate called after throwing an instance of 'std::out_of_range'
  what():  basic_string::at: __n (which is 10) >= this->size() (which is 10)
teletext: teletext.c:172: main: Assertion `ret == 0' failed.
./vb2: line 41: 14949 Aborted                 $HOME/vbit2/vbit2 --dir $PAGESDIRECTORY
     14950                       | $HOME/raspi-teletext-master/teletext -
VBIT2 Server crashed with exit code 0.  Respawning..

This then goes into an endless loop until i stop it with Ctrl-c.

peterkvt80, 2017-08-31:
This is being caused by vbit2 trying to read past the end of a string. It may be caused by the configuration file if you have one, or by a badly formed page file. Maybe send me MENU115.tti. It may be badly formed and I fail to trap the error.

nat2001, 2017-08-31:
Here it is. But it should'nt be any different to 104-124 as they are all made by the same code

nat2001, 2017-08-31:
For some reason Github won't upload the file...

nat2001, 2017-08-31:
DS,Inserter
SP,page11.html
DE,US &amp; Canada
CT,15,C
PN,11500
SC,1
PS,8040
MS,0
OL,0,P1xxCCX E1507 3108FNews story     AAuto
OL,1,—j#3kj#3kj#3k”“   |hh4|,|h<l4| h<l0
OL,2,—j $kj $kj 'k”“   ozz%pj7k4pjuz%
OL,3,—"###"###"###”/////-,,/,,,-.-.,,-,,/////
OL,22,„ƒHome news digest‡141ƒWorld digest‡142
OL,23,„ƒNews Index‡102ƒFlash‡150ƒRegional‡160
OL,24,Next News‚News IndxƒHeadlines†Main Menu
FL,116,102,101,100,F,100
OL,4,ƒ'Dangerous' chemical leak from Arkema
OL,5,ƒplant 
OL,6,‡A plume of smoke seen issuing from a
OL,7,‡flood-damaged chemical plant near the
OL,8,‡storm-hit US city of Houston is
OL,9,‡"incredibly dangerous", the head of the
OL,10,‡US emergency agency Fema has said.
OL,11,          
OL,12,†Brock Long said officials were still
OL,13,†trying to evaluate the situation at
OL,14,†Arkema's chemical plant in Crosby.
OL,16,†A police officer at the site was
OL,17,†hospitalised after inhaling fumes. Nine
OL,18,†others were admitted as a precaution.

ZXGuesser, 2017-08-31:
That's definitely been corrupted but I expect that's from pasting it into here as text with it being an 8-bit tti file

nat2001, 2017-08-31:
I tried to upload it but github won't let me for some reason

ZXGuesser, 2017-08-31:
you can email it to me if you like and I'll try. alistair@zxnet.co.uk

nat2001, 2017-08-31:
OK, but I don't know how this file could cause any problems

ZXGuesser, 2017-08-31:
nope, I can't attach it either, how odd. I expect it doesn't like the non-ascii characters. Anyway, I can't see anything wrong with it as you say.

ZXGuesser, 2017-08-31:
I've just tried to fire up vbit2 on my pi and it's freezing after a short while (not actually crashing though).

nat2001, 2017-08-31:
I've just noticed this line:
teletext: teletext.c:172: main: Assertion `ret == 0' failed.
Maybe the problem isn't with vbit but with raspi-teletext?

ZXGuesser, 2017-08-31:
try the code I just pushed. It's fixed my freeze.

nat2001, 2017-08-31:
No, it's still crashing...

[Packet830::Packet830] constructor
[~TTXPage] none
[~TTXPage] none
[Command::run] Newfor subtitle listener started
[Command::run] Ready for a client to connect
[FileMonitor::run] Monitoring /home/pi/Pages
[Service::worker]This is the worker process
[Service::run]Loop starts
[Service::run] No packet available for this line
[Service::run] No packet available for this line
[Service::run] No packet available for this line
[Service::run] No packet available for this line
[Service::run] No packet available for this line
[Service::run] No packet available for this line
[Service::run] No packet available for this line
[FileMonitor::run]  Adding a new pageP363.TTI
terminate called after throwing an instance of 'std::out_of_range'
  what():  basic_string::at: __n (which is 10) >= this->size() (which is 10)
teletext: teletext.c:172: main: Assertion `ret == 0' failed.
./vb2: line 41:  7096 Aborted                 $HOME/vbit2/vbit2 --dir $PAGESDIRECTORY
      7097                       | $HOME/raspi-teletext-master/teletext -

peterkvt80, 2017-08-31:
What the message is telling you is that vbit2 tried to read past the end of a string. It then terminates and then that causes raspi teletext to stop because its input pipe has gone away. If you could zip up your whole teletext pages folder then I could attempt to create the same conditions as yours. peterk.vt80@gmail,com

nat2001, 2017-08-31:
I've sent the pages

peterkvt80, 2017-08-31:
I got your pages, I had exactly the same problems as you. I can see what causes the problem. Two examples are MENU107.tti and MENU120.tti. In both of those the OL,10 line has ten blanks where I think it expects 40. This breaks the IsBlank() function.

nat2001, 2017-09-01:
OK, thanks. I'm updating my news code accordingly. (I cant remember why I wrote in that line anyway)

nat2001, 2017-09-01:
I removed this line, but it's still doing it.

nat2001, 2017-09-01:
It's crashing with Teefax as well as my own pages

nat2001, 2017-09-01:
OK I think I have it sorted now.

nat2001, 2017-09-01:
I'm sorry to have to re-open this, but it seems the problem continues. Do you think it could possibly have something to do with the pages not being in order?

peterkvt80, 2017-09-02:
Row order does not matter as vbit2 sorts these as it loads.  At the moment vbit2 is running with your page set and Teefax. There may be something missing from the Makefile. Try "make clean", then "make" and see if it comes to life.

nat2001, 2017-09-02:
OK, I updated it and it now works. Thanks for the updates! Since I'm here, I'd like to make a feature request: would it be possible to have vbit2 send the 'update/c8' control bit if a page changes? Obviously it would only need to be sent once.

nat2001, 2017-09-03:
I see that this has been implemented. Thanks!

closed 2017-09-03 by nat2001