peterkvt80/vbit2 #11: Possible security vulnerabilities

mpentler, 2018-02-02:
At the suggestion of Alistair Cree I’m logging this security issue which I think is worth mentioning.

The TCP connection set up to send subtitles over the network seems to listen on all IPs. It should really only be visible on the LAN. A lot of people have their PIs visible to the internet and a faulty router configuration could leave the system vulnerable.

There would of course need to be a vulnerability in VBIT2 itself to make this a problem, but it seems to be something worse closing off.

petergaal, 2018-02-03:
Did you mean listen to localhost? Because if it is listening on LAN network (which is connected to internet) I don't see how listening only on one interface would protect against getting access if the router will allow this access from outside.

mpentler, 2018-02-03:
You know, I think you might be right, I may have had a logic bypass here. Apologies. I saw it on a netstat and panicked.

ZXGuesser, 2018-02-03:
It would probably still be wise to make this default to being disabled and switch it on with a config file setting, since a lot of people still tend to put their Pi online without firewalling it properly. Then it would be an informed choice to open up the control port which most people aren't using anyway.

mpentler, 2018-02-03:
Also many won’t use the feature maybe?

peterkvt80, 2018-02-05:
Seems sensible to leave the network port switched off. If anyone has the technical ability to use the control port then editing the config file will be no problem to them.

ZXGuesser, 2018-02-05:
Ok, I've added a config setting `enable_command_port` and also replaced the `--port` argument to override the default with another config setting` command_port` (I expect I was the only person using that flag anyway as I never made a fuss that I'd added it)

closed 2018-02-05 by ZXGuesser