peterkvt80/vbit2 #12: accented characters?

wivaku, 2018-02-19:
Not really a vbit issue, but it is not clear to me how to use accented characters in .tti files.
At the moment I am creating UTF-8 files, with language set to German (`PS,8080`).
With `vbit2 | teletext` this results in garbled row.

nat2001, 2018-02-19:
There's information about this in the help files of wxTED. On P15 of the manual, there's a table for the keys

ZXGuesser, 2018-02-19:
You mustn't put unicode characters in tti files because they are interpreted as control codes. The tti file should contain the ascii characters which correspond to the desired German character in Table 36 of the teletext specification (ETS 300 706).

For German:
`# $ § Ä Ö Ü ^ _ ° ä ö ü ß`
``# $ @ [ \ ] ^ _ ` { | } ~``

wivaku, 2018-02-20:
Thanks ZXGuesser, indeed my mistake to put unicode in tti. 
Nat2001: I am not using wxTED, and not on Windows, but thanks for pointing me to [the manual](https://github.com/peterkvt80/wxted/blob/master/manual.odt)
This means: I set the .tti PS code to e.g. `PS,8080` for German and then `@` will appear as `§`?

Can I combine character sets? E.g. display `@` from English, plus `ä` from German and `ë` from French? 
Manual states: 

> Higher levels of teletext such as 2.5 and 3.5 add a region option.

Does vbit2 support these higher levels? If so, how what character code (the actual code in tti file) is used to specify the region?

ZXGuesser, 2018-02-20:
The region setting allows the selection of other languages and character sets beside the seven Western European ones, it can be set in a tti file with a RE line. e.g. `RE,0` is the default.

To display characters from different national option sets on the same page requires level 1.5 or higher page enhancements. vbit2 does support a lot of the higher level features now, but they're not really documented properly yet I'm afraid (It's a series of non-standard hacks I added to the tti file parsing).

My teletext editor (https://zxnet.co.uk/teletext/editor) supports higher level enhancements and can generate the appropriate tti files, however I only have a rudimentary utility for creating the enhancements and that also has no documentation...

wivaku, 2018-02-26:
Thanks ZXGuesser for the pointers and your TTI notes. The level 1.5 enhancements are probably outside my scope for now.
I have a related question (double height and newlines for Teletext subtitles), what is the best way to get in touch with you?

ZXGuesser, 2018-02-26:
If you are on facebook then joining the [Teletext group](https://www.facebook.com/groups/TeletextGroup) is a good idea for discussing general teletext stuff, I can then also explain stuff in instant message which is a bit quicker.

wivaku, 2018-02-26:
Done! What is your name on Facebook?

closed 2018-08-12 by ZXGuesser