peterkvt80/vbit2 #13: issues showing subpages in vbit2?

wivaku, 2018-02-20:
I create .tti files for pages that can have subpages.
The issue: they "flicker", where all / parts are shown multiple times a second. As if the cycle time is ignored and it is trying to show all pages at once.

In my .tti files I set the `PN,mppss` and also the Cycle info. E.g.:

```
DE,702-00 - 2018-02-20 23:03:34
PN,70200
CT,5,T
SC,0000
PS,8040
```
and subpage:
```
DE,702-01 - 2018-02-20 23:03:34
PN,70201
CT,5,T
SC,0001
PS,8040
```

The same for the standard sample page `R12400.TTIx` and `R12401.TTIx`.

ZXGuesser, 2018-02-21:
The subpages for a given page must all be contained within the same tti file.
For example something like this:
`
DE,702
CT,5,T
PN,70200
SC,0000
PS,8040
OL,1,some text rows
PN,70201
SC,0001
PS,8040
PS,1,some text rows
`

wivaku, 2018-02-21:
Ah, that explains it. Thanks! Now it works fine.

closed 2018-02-21 by wivaku