peterkvt80/vbit2 #14: Windows-precompiled vbit2.exe

veso266, 2018-06-15:
Hi I was wondering if you could compile vbit2 for windows and post it somewhere or tell me how to compile that properly on msys2 and g++ as compiler if possible

Thanks for Anwsering and Best Regards

ZXGuesser, 2018-06-15:
There's nothing particularly special as far as I can remember, just install msys2 as recommended, navigate to your vbit2 directory and run `make`

It does need to be a mingw shell, i.e. use the mingw32.exe or mingw64.exe, not msys2.exe

It uses the pthreads library if that's not installed by default you'll need to figure out what msys2 package it is and use pacman to install it.

veso266, 2018-06-15:
yep it compiled normally
PS: whats the diference between mingw32.exe or msys2.exe
always used mingw64.exe but never knew why (I can install stuff normally from MinGW64/32.exe not know what would I need to use msys2.exe anyway

PPS: is it possible to change vbit2 directory where it looks for ttx files (because it always wants J:/Temp/teletext (not sure where it picked my external drive but its not alone because Net Framework installer dump its crap there too)

closed 2018-08-12 by ZXGuesser