peterkvt80/vbit2 #15: dead links, hosting and dailyprog

mabynogy, 2018-07-04:
I noticed many links are broken. It seems you've reach the bandwidth quota.

We are a bunch of programmers chatting and hacking together. Some of us are interested by teletext. We  have a server and we can host some parts or your project if you want.

Feel free to reach us on http://dailyprog.org/chat/ and/or to delete that issue.

Cheers

peterkvt80, 2018-07-05:
Thanks for that. Something blew up the usage last weekend. My legacy hosting doesn't have any safeguards for that. They just archive the site. I am getting fibre broadband and I was planning to run a Raspberry Pi server through that. Your solution might be better.

mabynogy, 2018-07-05:
You can do both and embed one or another in an iframe to keep the current url (if you prefer). Tell me the ssh username you wish. My email is m@ the same domain name.

ZXGuesser, 2018-07-05:
I've been running all my sites off my home broadband for years with no issues so far touch wood! Just have to keep on top of updates and lock everything down hard!

ptdel, 2018-07-05:
@peterkvt80 I hope me poking around on the :8080 instance wasn't causing issues. It's great that you are making this stuff available for people who are interested.  I was able to compile vbit2 just fine on archlinux and have it stream to stdout with `~/vbit2 2>/dev/null`. @ZXGuesser would your *teletext-decoder* be something I could view the output of vbit2 with?  I am still trying to find a client that doesn't require raspberry pi libraries to build.

ZXGuesser, 2018-07-05:
teletext-decoder is for analysis of teletext dumps it's not intended for reading teletext. It can be used fine with vbit2 output however I found I need a small script that consumes packets from vbit2 at the appropriate rate and dumps them to a file otherwise things go a bit bonkers.

peterkvt80, 2018-07-05:
The 8080 port is just a Raspberry Pi sitting on my fireplace. It is not connected with my teastop.co.uk webspace. I normally get about 300 hits per month. I got 30,000 hits in one day. I have taken up Mabynogy's kind offer and my webpages can now be found at http://dailyprog.org/~peterkvt80/. There is nothing significant changed except for the URL.

closed 2018-07-05 by peterkvt80