peterkvt80/vbit2 #18: Teletext-Output to OpenCaster?

premultiply, 2018-12-11:
Have not tried it yet but should it be possible to send VBIT2s stream directly to OpenCaster to packetize and mux it into a DVB service?

peterkvt80, 2018-12-11:
I'm sure it should be possible but I don't know OpenCaster so I can't say exactly how. You might need to adapt the VBIT2 output. Possibly remove some bytes from the start of a packet, like the clock run-in and WST sync byte. Also the --reverse option could come in useful if the bit order is wrong. If it needs more adapting then a little Python could do the processing. The idea is to get it into a shape where you can run it through txt2pes etc.

premultiply, 2018-12-11:
Oh yes, it expects 46 bytes for each line.
Another idea would be to add an option to VBIT2 to just leave out the additional fixed starting bytes.

peterkvt80, 2018-12-11:
It isn't difficult to do changes to the format of VBIT2's output. At the moment it uses 16 lines per field and 42 bytes per packet. If you want the two clock run ins and framing code you can add that to the start of each packet. The vbit-py project does exactly that.  This is how it adds clockFrame = b'\xaa\xaa\xe4'. It gets run by this script: /home/pi/vbit2/vbit2 --reverse 2> /dev/null | /home/pi/vbit-py/vbit.py

premultiply, 2018-12-11:
I read in https://www.etsi.org/deliver/etsi_en/300400_300499/300472/01.04.01_20/en_300472v010401a.pdf Section 4.3 that it would be a little bit more complicated because the field_partity and line_offset has to be included dynamicly. data_unit_id may be dynamic too (for perfection synced to subtile flag). All other bytes are static.
So that has to be generated by VBIT2 as only it has knowledge about it.

ZXGuesser, 2018-12-11:
VBIT2 can do other numbers of lines per field now using the lines_per_field option in its config file (same value used for both fields)

For the PES data field I thing you're best to count lines.

closed 2020-01-12 by ZXGuesser

SonnyWalkman, 2021-01-14:
Exactly what I need using opencaster.