peterkvt80/vbit2 #19: Support for ETT (.ttp) file format

premultiply, 2018-12-13:
Maybe you are interested in adding support for the FAB ETT file format.
This is the format description:

```
	When teletext pages are stored in FAB ETT file format the files are stored with extension .TTP and contain following data:

	Byte 0:				Number of used rows (1 to 49)

	Byte 1..42:	Row 0
		Byte 1:			SO (0Eh)
		Byte 2:			80h
		Byte 3...32:	Page name + odd parity (first 30 bytes, rest starts at position 2060)
		Byte 33..35:	80h
		Byte 36:		Left Margin (0 to 38) + odd parity
		Byte 37:		Right Margin (1 to 39) + odd parity
		Byte 38:		Top Margin (1 to 22) + odd parity
		Byte 39:		Bottom Margin (2 to 23) + odd parity

		Byte 40:		Time for display (3 to 60), bit with value 64 should be set to disable numbering of subpages (FAB System only)+ odd parity

		Byte 41:
			bit 0 (1) 	= C7, header suppress
			bit 1 (2)	= C4, erase page
			bit 2 (4)	= C9, interrupted sequence
			bit 3 (8)	= C10, inhibit display
			bit 4 (16)	= C5, newsflash
			bit 5 (32)	= C6, subtitle
			bit 6 (64)	= set to 0
			bit 7 (128)	= odd parity

		Byte 42:		Character Set + odd parity	(0=GB, 4=D, 2=S, 6=I, 1=F, 5=E, 3=CS)

	Byte 43..2059:		Data rows (as specified in byte 0 minus 1), each consisting of block of 42 bytes:
		Byte 0:			SO (0Eh)
		Byte 1:			Packet (Row) number + odd parity
		Byte 2..41:		Contents of data line

	Byte 2060..2069:	Page name + odd parity (last 10 bytes)

	Remarks:
	1. Data lines should be stored in the following order: X/27 (0 or 1 packets),  X/26 (0 to 15 packets),  X/1..X/23 (only lines which contain characters other than 20h), X/24 (0 or 1 packet), X/25 (0 or 1 packet)
	2. Data in data lines should be encoded according to teletext specification. For lines X/1 to X/23 use odd parity characters.
	3. Unused space at the end of file can be filled up with character 20h.
```

Each page file has a size of 2070 bytes. I can provide some sample files.

Here is some sample PHP code to read a ETT file:
```
	public function loadETT($filename) {
		if (filesize($filename) == 2070) {
			$fh = fopen($filename, 'rb');

			//fseek($fh, 0);
			//print "Number of used content rows: ".ord(fread($fh, 1))."\r\n";

			fseek($fh, 3);
			$this->PageContent[0] = $this->ParityStrip(fread($fh, 30));

			fseek($fh, 40);
			$this->Duration = ord(fread($fh, 1)) & 0x7f;

			fseek($fh, 41);
			$flags = ord(fread($fh, 1));
			$this->FlagInhibitHeader  = $this->TestBit($flags, 0);
			$this->FlagErasePage      = $this->TestBit($flags, 1);
			$this->FlagOutOfSequence  = $this->TestBit($flags, 2);
			$this->FlagInhibitDisplay = $this->TestBit($flags, 3);
			$this->FlagNewsflash      = $this->TestBit($flags, 4);
			$this->FlagSubtitle       = $this->TestBit($flags, 5);

			fseek($fh, 42);
			$this->CharacterSet = ord(fread($fh, 1)) & 0x7f;

			for ($i = 0; $i < 48; $i++) {
				fseek($fh, ($i) * 42 + 44);
				$rownum = ord(fread($fh, 1)) & 0x7f;

				fseek($fh, ($i) * 42 + 45);
				$rowdata = fread($fh, TTX_LINELENGTH);
				if ($rownum < 26) {
					$this->PageContent[$rownum] = $this->ParityStrip($rowdata);
					if ($rownum == 24) $this->extractX24($this->PageContent[$rownum], $this->char_replacement_map[$i]);
				}
				else {
					$this->PageExtension[$rownum][] = $rowdata;
					switch ($rownum) {
						case 26:
							$this->decodeX26($rowdata);
							break;
						case 27:
							$this->decodeX27($rowdata);
							break;
					}
				}
			}

			fseek($fh, 2060);
			$this->PageContent[0] .= $this->ParityStrip(fread($fh, 10));

			fclose($fh);
		}
	}
```

Page number and Subpage number has to be parsed from the filename (like 100_00.ttp) or may be written to Row 0 as text like "100.00 Some Page Title here".

There is also a variant of this file format to support multiple subpages in one file but I did not have a look at it and do not have a format description yet.

ZXGuesser, 2018-12-13:
There are some vestigial bits of code for other file formats already in the codebase which ought to be made functional or removed. I'm not sure whether it's better to try to make vbit2 support a whole bunch of formats, or to write a converter to bulk convert other formats to tti.
Since vbit2 gets things like carousel cycle timings etc from the tti files importing other formats would leave you stuck with default values for any settings the format can't hold.

ZXGuesser, 2018-12-13:
As an aside to this, I have been meaning to add ETT to my teletext editor but not got around to it. If you have a licensed copy of FAB that you could make certain test files for me that would be helpful!

premultiply, 2018-12-13:
Missing properties in some formats can be easily parsed from the filenames for these cases.
Something like "PAGE_SUBPAGE_TIMING Some text.EXT". This is like some old commercial systems does it.

Yes, I have lots of these FAB files ;-)

peterkvt80, 2018-12-13:
I'll certainly put FAB support into wxTED. TTI was never designed with higher level teletext in mind although it is an easy format to add features to. If there is a better file format then I'd consider adding it.  As for FAB, I wonder how much it is to get a one year license for the basic package?

ZXGuesser, 2018-12-13:
yeah, this is the problem with adding stuff to vbit2 though, it's all very tied to each file being an entirely separate entity and I suspect would take quite a lot of reworking to support things like multiple files per page and metadata from separate files*. Not saying that that might not be a more flexible system, but it would be a lot of work compared to something that parses a directory of such files and generates the appropriate tti files from them.

* (If going down that road it would probably be better to start again and rewrite the entire thing to store pages in a database rather than a directory of files in a bunch of random formats anyway)

premultiply, 2018-12-13:
I only own the old ETTWIN FAB Teletext Editor with USB dongle. Most part of it is 16 bit... I am not shure if it is still sold. ;-)

The file folder system has many advantages over complex and heavy database systems.
It is fast and efficient. Just drop, update or delete something there and teletext will be updated. Just run a few cronjobs to update the data. I wrote a large library to fully automate teletext page generation and full decoding (text console and web) with many features like text boxing, alignment, line breaking, formating, charset conversion, .... It just reads and write files from folder in the requested teletext file formats.