peterkvt80/vbit2 #22: Static Bar at The Top of the Screen

KubbaYioka, 2019-07-11:
Hello,

I'm using a RPI 3B+ with latest Raspbian Stretch installed. I'm getting a bar of static at the top of the screen when I run Teefax, and no image. I can only see the desktop.

Does this program work with NTSC televisions since sdtv must be set to 2 (which doesn't work on my set)?

peterkvt80, 2019-07-11:
Sounds like it is working correctly. vbit must be PAL, sdtv mode 2 with composite video (not HDMI) and a teletext TV. I have not tested NTSC with vbit. In theory it could work but vbit is only set up for PAL timing so it would need some changes. Some USA TV sets support PAL and teletext but if you don't have teletext buttons on your remote control then it definitely does not support teletext.

KubbaYioka, 2019-07-18:
I don't think Vbit is an option for my current setup. Do you know of any similar programs out there that can do something similar?

peterkvt80, 2019-07-19:
VBIT is a teletext inserter so it only generates the vbi over the video out. This is what looks like static. You can only view the pages with a TV that provides decoding and display. What it seems you expected was that vbit is an in-vision system but it isn't. You need is a teletext viewer that simulates a TV. I'm not sure what you want to do but there are viewers https://temp.zxnet.co.uk/teletext-viewer/?channel=1&page=520.

KubbaYioka, 2019-07-23:
You have been very helpful! Yes, that is closer to what I am searching for. Honestly, I'm really just searching for something that can scroll news and information in a minimalist format similar to what teletext did for people decades back. Weatherstar 4000+ comes to mind. 

If you have any additional insight into what I might be looking for, I'd be happy to hear it. In any case, I think we can close this issue.

Thanks again for taking the time to help!

nat2001, 2019-07-24:
How about [this](https://internal.nathanmediaservices.co.uk/textbulletin/) lol

closed 2020-01-12 by ZXGuesser