peterkvt80/vbit2 #24: Using my own page server

bartleph, 2019-10-31:
Hi Peter. Great article and software. Thank you for sharing. I am hoping to use my own page server with this. I have a Subversion server on my local network with modified pages set up. Is it just a case of changing the destination in the getvbit2 file to point to the new repo?
Thanks.

ZXGuesser, 2019-10-31:
That would work, or you could write your own simpler install script (that doesn't download the unneeded pages.zip for example).
The page directory is specified in the `--dir` argument when vbit2 is started. If you already have vbit2 installed through getvbit2.sh, the page directory is set in the `vb2` script by the `PAGESDIRECTORY` variable.

bartleph, 2019-11-01:
Many thanks for that. I've managed to get it running to my own Subversion server.
Kind Regards.

closed 2019-11-01 by bartleph

MICR-OSOFT, 2021-03-31:
I find this interesting, But could I get a step-by-step guide on a fresh install of vbit2?

peterkvt80, 2021-03-31:
This is off topic and covered by the Wiki, but these two videos are great guides. Detailed technical look with a few peeks under the hood. NiksCave [teletext install](https://www.youtube.com/watch?v=PQLhmKh1qI4)
Shorter and entertaining guide for absolute beginners. Tomodachisim534 [teletext install](https://youtu.be/t8CNr6-9620)