peterkvt80/vbit2 #27: Remove redundant code

ZXGuesser, 2020-01-12:
Purge irrelevant code carried over from editor.

- Page saving is used in commented out debug code in SendSubtitle is this needed, would a simpler page dump suffice for debugging?
- There is code for multiple formats but we only read tti. Do we want to (re)implement loading the other formats or should we throw them out? Affects https://github.com/peterkvt80/vbit2/issues/19

peterkvt80, 2020-01-13:
Probably best to drop the irrelevant code. So get rid of page saving and formats other than tti..

ZXGuesser, 2020-01-15:
That's chopped out a bunch of dead stuff in 0f6b2bbdd146c5820ba2a47ee9700ec731e62e17. There's no doubt more in code that I don't fully understand.
I notice there's duplication of page copying, could this be consolidated? I see that it's used by the subtitle stuff which I've never dug into enough to figure out how it all works.