peterkvt80/vbit2 #30: Using the vbit config

ZombieWorm, 2020-12-22:
Hi gents hoping I can get round this. When using the config tool I get this error "svn checkout failed with error 1". Is this an issue I can get help with?

TIA

ZXGuesser, 2020-12-22:
Which service?

ZombieWorm, 2020-12-22:
Apologies ZX what is the best method to do this debian is not my strong point?

ZXGuesser, 2020-12-22:
I meant using the remove option in vbit-config if the service is showing there but broken. If it can't check out the service on a blank system I'm a bit puzzled because both the Teefax and Ceefax services which use svn seem online and fine from here.
You can check if there's anything in or delete the whole hidden .teletext-services directory in your home directory to manually clear out corrupted services.

ZombieWorm, 2020-12-22:
The config only offers install (which leads to the error). I manually deleted the hidden directory you mentioned then tried to use the config again but it leads to the same error..

Must be something beyond my scope as been using pi for years, I have four pi2's and two original pi's - only thing I have not ruled out is another service like samba using same server settings as it is the only thing on a virgin install I changed when connecting up a disk drive to use as a NAS?

avrovulcanxh607, 2020-12-24:
Have just tried this on a clean install myself and seems to be working fine, which rules out a server-side issue. The only thing I can suggest is to try checking out Teefax or Ceefax with SVN manually to see if that works.

peterkvt80, 2020-12-24:
It isn't unheard of for the Teefax server or my Plusnet broadband to go wrong, It is a Pi 1 that has been running since 2015. It has trashed one SD card and it must be due for another failure any day now. At the moment it seems fine. The web access to Teefax is at [http://teastop.plus.com/svn/teletext/](http://teastop.plus.com/svn/teletext/ ) and if you can access that then it is probably OK.

closed 2023-01-16 by ZXGuesser