peterkvt80/vbit2 #31: Teletext into DVB.

SonnyWalkman, 2021-01-14:
Not sure if the raw teletext produced by vbit2 can be somehow injected into  pes? I guess old to new format?

premultiply, 2021-01-14:
Yes, possible. See #18.

SonnyWalkman, 2021-01-24:
> Yes, possible. See #18.

Thank you, I believe you have done what is required. I check it out.

SonnyWalkman, 2021-10-11:
Another left field question. Now gstreamer mpegtsmux 2.19. X is close to muxing a dvb compatible TS, could vbit code be sandwiched into a gstreamer plug-in to use with encoding a teletext stream with mpegtsmux using the application/x-teletext caps?

Something like:-

gst-launch-1.0 teletextenc pages=<some filepath> ! application/x-teletext ! mux.sink_1234 videotestsrc ! is-live=true ! video/x-raw, width=1280, height=720, framerate=25/1 ! mpegtsmux name=mux ! filesink location=my test.ts

premultiply, 2021-10-15:
I think it would be best to make vbit2 generate a PES stream (without PTS).

ZXGuesser, 2021-10-15:
There's slightly more to it than just changing what's written to the output as it needs to generate the PES headers and append the right number of lines worth of data to each PES packet to match the size needed by the transport stream, and use the line and field counters to build the data field. All the required information is available in `Service` though.
I would wrap it all in a new function to call with the packet data instead of the `cout.write`, then it can be easily put behind a command line switch like the byte reversal.
The other minor issue is that EBU teletext data in PES only supports 16 lines per field. This currently doesn't matter as none of the full-field configuration has actually been plumbed in yet, but the line count variable could hold values greater than 15 in future so that should be handled appropriately by the code generating `line_offset` in the data field.

ZXGuesser, 2021-10-15:
Thinking about it, there's a much more fundamental issue. As far as I can recall there's nothing in vbit2 to actually regulate the rate it generates packets other than blocking until the buffer gets emptied. This works so long as you're consuming the right number of lines worth of packet data every 20ms but otherwise it goes bananas and spews thousands of packets per second.
This is something I've never been happy with, and should probably be fixed before doing anything else.

ZXGuesser, 2021-10-15:
The problem is that besides the process receiving the data locating and parsing vbit2's config files, there's currently no way to determine how much data you need to consume each frame, which is why the scripts for using vbit2 with raspi-teletext are so ugly!

ZXGuesser, 2021-10-18:
I don't really understand the question :)

Just updating to say I made some progress on the stream timing stuff yesterday, but still have some loose ends to tie up and still need to test it thoroughly with raspi-teletext to prove I still have accurate output without buffer underruns etc.
Once that's done I can start altering the output routine to allow a PES mode.

ZXGuesser, 2021-11-26:
I've implemented basic PES output in 290cf7ca0ad52cb89da94e94a5886fa73be9fb13

It's enabled via `--format PES` on the command line. I have no way to mux to a transport stream to test it, but extracting teletext data from the resulting PES gives the expected data back.

There's currently no Presentation Time Stamps

ZXGuesser, 2021-12-05:
Has anyone been able to test this yet? @premultiply @SonnyWalkman

premultiply, 2021-12-05:
Not yet, sorry. I am a bit busy at the moment with other things but I hope I have some time for it in the next days.

ZXGuesser, 2021-12-05:
No worries. It took me weeks to get around to writing it after all! I just wanted.to check it'd been seen 🙂

premultiply, 2022-02-04:
Works fine for the first look after fixing the data_unit_length to 0x2c.
But each field should be separated into its own PES packet for better compatibility.

ZXGuesser, 2022-02-04:
Ok, I'll look at separating the fields. I'm sure there was a good reason I did it that way, but I've forgotten now! 😄

ZXGuesser, 2022-02-05:
Does https://github.com/peterkvt80/vbit2/commit/ae91e6199f8cc88af917c9e5ba407e2647f88dbd give the expected output?

premultiply, 2022-02-06:
Yes, perfect!

ZXGuesser, 2022-02-06:
Great stuff, I'll get that merged then. 👍
Any chance of an idiot's guide to generating a working transport stream with it that @avrovulcanxh607 and I can try ourselves? 🙂

premultiply, 2022-02-06:
Great work!

Yes, of course. I do it with a simple script using the OpenCaster toolset to generate a clean teletext-only-TS including all tables with a vbit-config generating 11 lines per field (3 PES packets per field):

```shell
#!/bin/sh

TS_BITRATE=1504000

TTX_BITRATE=$((3 * 188 * 50 * 8))
TTX_PID=257

mkfifo txt.ts
mkfifo mux.ts

/opt/vbit2/vbit2 --dir "/opt/teletext" --format PES | pesdata2ts /dev/stdin $TTX_PID > txt.ts &

tscbrmuxer b:15040 pat.ts b:15040 txtpmt.ts b:752 sdt.ts b:151 nit.ts b:$TTX_BITRATE txt.ts o:$TS_BITRATE null.ts > mux.ts &
tsudpsend mux.ts 192.168.0.100 1234 $TS_BITRATE
```

closed 2022-02-06 by ZXGuesser

SonnyWalkman, 2022-02-06:
Thanks for all the effort vbit2 and petervt80 for making PES support
happen. Im yet to try it out.. Have out of action for a while.

I’ll looking to try on my software DATV repeater which uses Gstreamer as
the core pulled together with python.
Hoping to rewrite in rust once concept is working.

Rob VK3EHT