peterkvt80/vbit2 #34: Stream timing related improvements

ZXGuesser, 2021-10-15:
- [ ] Regulate packet generation to internal master clock rather than relying on output buffer blocking.

Possible future enhancements:
- [ ] Migrate all time related functions to use internal master clock synchronised to field rate and bsdp generation.
- [ ] Build packet buffer forwards into future time.

Affects #31

SonnyWalkman, 2021-10-15:
Sounds good.
This does allow generating a timed  PES packet easier to use with DVB
transport streams.

ZXGuesser, 2021-11-03:
Have made progress on this over the past week and while I still have some testing to do, it looks good so far.
I can now begin tidying up the packet output code to work towards supporting different packet output formats such as PES, via command line switches.

SonnyWalkman, 2021-11-03:
Great work ZXGuesser!!