peterkvt80/vbit2 #35: TS encapsulation of Teletext PES for tsduck

omikron88, 2023-02-24:
I made some tests with including teletext packets to tsduck live streams
This worked for me: (written in Python)

`
def send_ts():
    global ts, pid, pes, \
        ts_payload, ts_index, ts_cc, \
        TS_PACKET_SIZE, TS_HEADER_SIZE
    temp = TS_PACKET_SIZE - TS_HEADER_SIZE
    if ts_payload >= temp:
        ts[3] = ts_cc | 0x10  # continuity counter, no scrambling, only payload
        ts_cc = (ts_cc + 1) % 0x10  # inc.continuity counter
        ts[TS_HEADER_SIZE:] = pes[ts_index:ts_index + temp]
        sys.stdout.buffer.write(ts)
        # out.write(ts)
        ts_payload -= (TS_PACKET_SIZE - TS_HEADER_SIZE)
        ts_index += (TS_PACKET_SIZE - TS_HEADER_SIZE)
        ts[1] = (pid >> 8) & 0x1F  # pid not payload unit start
        ts[2] = (pid & 0xFF)
        if ts_payload != 0:
            send_ts()
    else:
        ts[3] = ts_cc | 0x30  # continuity counter, no scrambling, adaptation field and payload
        ts_cc = (ts_cc + 1) % 0x10  # inc.continuity counter
        ts[TS_PACKET_SIZE - ts_payload:] = pes[ts_index:ts_index + ts_payload]
        ptr = TS_PACKET_SIZE - ts_payload  # point to the first payload byte
        ts[4] = ptr - TS_HEADER_SIZE -1
        ts[5] = 0x00  # no options
        for temp in range(6, ptr):
            ts[temp] = 0xFF
        sys.stdout.buffer.write(ts)
        # out.write(ts)
        ts_payload = 0

def init_ts():
    global pid, ts_payload, ts_cc, \
        TS_PACKET_SIZE
    ts = bytearray(b'\xff') * TS_PACKET_SIZE
    ts_payload = 0
    ts_cc = 0
    return ts

def add_ts(size):
    global pid, ts_payload, ts_index
    ts_payload = size
    ts_index = 0
    ts[0] = 0x47  # sync
    ts[1] = (pid >> 8) & 0x1F  # pid
    ts[2] = (pid & 0xFF)
    ts[1] |= 0x40  # payload unit start indicator
    send_ts()

def init_pes(pes_size):
    global packet_index, EBU_UNIT_SIZE
    pes = bytearray(b'\xff') * pes_size
    pes[0] = 0x00
    pes[1] = 0x00
    pes[2] = 0x01  # prefix
    pes[3] = 0xBD  # data
    temp = pes_size - 6
    pes[4] = (temp >> 16) & 0xFF
    pes[5] = (temp & 0xFF)
    pes[6] = 0x85
    pes[7] = 0x80  # flags
    pes[8] = 0x24  # header size
    # 31 0xFF stuffing here
    pes[45] = 0x10  # ebu teletext
    packet_index = EBU_UNIT_SIZE
    return pes

def stamp_pts(stamp):
    global pes
    pes[9] = ((stamp >> 29) & 0x0F) | 0x21
    pes[10] = (stamp >> 22) & 0xFF
    pes[11] = ((stamp >> 14) & 0xFF) | 0x01
    pes[12] = (stamp >> 7) & 0xFF
    pes[13] = ((stamp << 1) & 0xFF) | 0x01

def add_pes(buffer):
    global pes, packet_index, pes_size, \
        pts_stamp, pts_increment, \
        EBU_UNIT_SIZE
    pes[packet_index:] = buffer[0:]
    packet_index += EBU_UNIT_SIZE
    if packet_index == pes_size:
        stamp_pts(pts_stamp)
        add_ts(pes_size)
        pts_stamp += pts_increment
        packet_index = EBU_UNIT_SIZE

`
The PES packets need to include PTS timestamp. Because tsduck merge plugin controls the insert rate. pts_increment is 1800 for 50Hz time advance, or 3600 for 25Hz time advance. It is in 90kHz units.
Next goes the encapsulation to 188 byte TS packets. pid is the required PID from 0x0100 to 0x1FFE.
It is possible to inject the resulting packets into live tsduck steam with acceptable time error:
-P merge "gener .... "

Of course, there is also need to add the reqired signalisation to PSI tables:
ttx.xml:
`
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <PMT>
    <component elementary_PID="1024" stream_type="0x06" x-node="add">
      <VBI_data_descriptor>
        <service data_service_id="1">
          <field field_parity="true" line_offset="8"/>
          <field field_parity="false" line_offset="8"/>
        </service>
      </VBI_data_descriptor>
      <teletext_descriptor>
        <teletext language_code="cze" teletext_type="0x01" page_number="898"/>
      </teletext_descriptor>
      <stream_identifier_descriptor component_tag="0x07"/>
    </component>
  </PMT>
</tsduck>
`
edit your language, start page, PID and VBI lines used. And patch the PMT table by
-P pmt --patch-xml ttx.xml

Have a nice day

omikron88, 2023-02-24:
Here are the source files
[source.zip](https://github.com/peterkvt80/vbit2/files/10828393/source.zip)

SonnyWalkman, 2023-02-24:
Great work,

Indentation is out. any chance to add this snippet in a test folder under
TS duck?

Rob

SonnyWalkman, 2023-07-15:
I'm just beginning to play with getting vbit2 to send to TSduck.
Do you have a working TSduck implementation? 
Furthermore, can vbit2 pass T42 using UDP into TSduck?

SonnyWalkman, 2023-07-15:
I found you bash file.. I got Teletext working with VLC however the PCR's is way high.
ffmpeg is encoding and supplying a 4.5M CBR rate TS stream over multicast socket local on the machine.

* pcrverify: PID 256 (0x0100), PCR jitter: -4,458,984 = 165,147 micro-seconds = 52 packets + 132 bytes + 6 bits (tsp time)

My script below, anything which l could improve?

tsp -v -b 480000 ^
-I ip 224.0.0.1:9999 ^
-P regulate  ^
-P merge -b 4700 "python ttxgen.py ttx.t42 2 0 3600 1024" ^
-P pmt --patch-xml TTX.xml ^
-P inject -p 16 NIT.xml=2000 ^
-P inject -p 20 TDT.xml=1000 ^
-P nit --build-service-list-descriptors ^
-P pcrverify ^
-O ip 224.0.0.2:9999

SonnyWalkman, 2023-07-16:
The above method kind of worked however, I have heaps of PCR discontinuity errors. Video stopping. lost frames etc.
I've thought about it some more.
I think the better approach is create a completely new TS with its own PCR timing and have that muxed in with the live AV TS?

TSduck has the capacity to manufacture a TS from the ground up and just add all what is need to make the stream compliant. Taken from the TSduck documentation.

BITRATE=400000
PCR_PER_SEC=5
PCR_DISTANCE=$(( $BITRATE / $(( $PCR_PER_SEC * 188 * 8 )) ))
PCR_PID=5004
tsp --verbose --bitrate $BITRATE --max-flushed-packets 70 ^
-I null ^ 
-P regulate --packet-burst 14 ^
-P filter --every $PCR_DISTANCE --set-label 1 ^
-P craft --only-label 1 --pid $PCR_PID --no-payload --pcr 0 ^
-P continuity --pid $PCR_PID --fix ^
-P pcradjust --pid $PCR_PID ^
-P inject PAT.xml --pid 0 --bitrate 15000 ^
-P inject SDT.xml --pid 17 --bitrate 15000 ^
-P merge -b 4700 "python ttxgen.py ttx.t42 2 0 3600 1024" ^
-P pmt --patch-xml TTX.xml ^
-P inject -p 16 NIT.xml=2000 ^
-P inject -p 20 TDT.xml=1000 ^
-P nit --build-service-list-descriptors ^
-P pcrverify ^
-O ip 224.0.0.3:4500 --packet-burst 7 --enforce-burst 

What do you think?

SonnyWalkman, 2023-10-29:
Hello @omikron88. 

Thanks for your contribution. 
It's been sometime and I can confirm your Python generator work beautifully with TSduck. 

I'm converting your python code to Go code to expand further in generation of both a 4-6 page carousel to cycle a static and a few dynamically generated pages to further provide a live teletext service. I'd like to understand generating the t42 stream to do such? I'll need to update the date and time in the header and developed pages to show system stats such as cpu and gpu, ram usage etc. 

I like the idea of a teletext fish tank however working out how I can replicate such in code require some thought.

https://youtu.be/zFkBHuchOnk?si=utewe3efCCaLMgtL

SonnyWalkman

peterkvt80, 2023-10-30:
It doesn't sound too difficult to generate the fish tank. You could adapt code from vbit2 as that already generates t42 and has everything needed to generate packets and sequence them correctly.

SonnyWalkman, 2023-10-30:
I’m coding in Go there looking at a small device to produce a index with a
few dynamic pages like weather, club news and a few engineering pages like
test charts and of course fish tank.

I have an idea using asciiaquarium however it written in Perl and leverages
off a few packages to work.

The concept is all there however, structuring fish and other objects in
teletext block art is a skill.  At 40x25 lines

Regards

Rob aka SonnyWalkman

Sent from Gmail Mobile iPhone 14

ZXGuesser, 2023-11-11:
Following on from a discussion on the facebook teletext group, I've hacked together some code to generate a PTS and encapsulate the PES output in a transport stream in a new branch: 1c97af2623a35e8a6a2a162009693f32242f4d32. It builds a PCR and PTS from the internal field count which tsduck can then resynchronize to another stream.
I've been able to successfully merge this into a live transport stream with tsduck.

SonnyWalkman, 2023-11-11:
Wow that was quick. I’ll get if a shot and send you feedback.

ZXGuesser, I appreciate your time and motivation in adding PTS time
stamping and TS encapsulation.

ZXGuesser, 2023-11-11:
I got thoroughly nerdsniped on this one, but the chances are I wasn't going to get anything else done today anyway 😂

ZXGuesser, 2023-11-16:
The latest code adds a `--pid` argument and renames the format from `PES` to `ts` to reflect that it's now encapsulated.
Before I merge this and do a release, is this new arrangement suitable for your uses @premultiply?

ZXGuesser, 2023-11-30:
Made some more small tweaks to correct the TS continuity counters, so tsanalyze doesn't report a zillion continuity errors now.
Merging the teletext stream into an existing transport stream seems to work, provided the target service is resynchronised to the new teletext pid (in tsp `-P svresync -p <teletext pid> <service id>`). This seems rather a brute force method but there doesn't seem to be any way to re-time the merged pid itself to a reference PCR with tsduck.

I think it's ready to merge but would appreciate more testing, particularly against real DVB hardware, before I release.

premultiply, 2023-11-30:
I think Teletext PID without PTS may be the preferred way here.

ZXGuesser, 2023-11-30:
Certainly that was my feeling, that it seems like every teletext decoder is happy ignoring the lack of a PTS, but @SonnyWalkman wanted it for some reason! 😆
I ought to be able to add another config flag to make it optional without too much difficulty. Reading in a reference stream to synchronize to is too complicated in my view and means adding loads more mpeg gubbins to vbit2

ZXGuesser, 2023-12-03:
Added another format `tsnpts` to suppress the PCR and PTS 438b9a5d625ff9d81d85fbdc7777b924048cd086

SonnyWalkman, 2023-12-04:
Hello @ZXGuesser and @premultiply,
Thank you for the efforts in adding the code to encapsulate the PES into a DVB TS frame.  

>  @SonnyWalkman wanted it for some reason! 😆

The reasoning behind the PTS and PCR functionality was to acknowledge the DVB standard in providing a PTS to cater for device requiring a PTS to be available to show teletext.

@ZXGuesser,

Is there a chance to add my fix to stop VBIT shutting down the remote connection if an error is received? 

I've added a fix however, @peterkvt80 may not be available to look at the issues with commands returning errors.

premultiply, 2023-12-04:
> The reasoning behind the PTS and PCR functionality was to acknowledge the DVB standard in providing a PTS to cater for device requiring a PTS even though the device could disregard the PTS altogether.

I have not found any device with a requirement for PTS on TXT PID yet. They all ignore it completly even if it exists.
Some public broadcasters sent out async PTS on their "HD channels" (simple PID-copy from "SD channel") for years. Nobody cared about it.

premultiply, 2023-12-07:
Yep, from todays all-IP practical playout system design it is not easy to mux a teletext PID with a **valid** PTS as it was in the analog or digital SD days.

**IF** you (still) have an encoder supporting teletext at all, most of the times this requires a teletext insertion into SDI using SMPTE2031 or OP-47 insertion and transport at the input of each encoding device for each tv service.

But most of the audio+video encoding systems do not have any support for teletext so the only way to go is to add it afterwards by muxing the teletext PID (without PTS) into the the "main" TS with video and audio.

ZXGuesser, 2023-12-09:
Released as v2.6.0

closed 2023-12-09 by ZXGuesser