peterkvt80/vbit2 #36: vbit-config traceback when adding "Directory" as a custom service

gkthemac, 2023-08-27:
When trying to add a custom service from a directory, vbit-config exits unexpectedly with a traceback.

OS: Raspberry Pi OS 11 "bullseye" 32-bit

Steps to reproduce:
1. Start vbit-config
2. Select "Install service"
3. Select "Custom service"
4. Select "Directory"

Actual result:

```
Traceback (most recent call last):
  File "/home/gkm/.local/bin/vbit-config", line 289, in <module>
    mainMenu()
  File "/home/gkm/.local/bin/vbit-config", line 273, in mainMenu
    installServiceMenu(servicesData)
  File "/home/gkm/.local/bin/vbit-config", line 54, in installServiceMenu
    customServiceMenu()
  File "/home/gkm/.local/bin/vbit-config", line 129, in customServiceMenu
    code, string = d.dselect(os.getenv('HOME')+"/", title="Enter teletext service directory:", height=8)
  File "/usr/lib/python3/dist-packages/dialog.py", line 2363, in dselect
    return self._widget_with_string_output(
  File "/usr/lib/python3/dist-packages/dialog.py", line 1719, in _widget_with_string_output
    code, output = self._perform(args, **kwargs)
  File "/usr/lib/python3/dist-packages/dialog.py", line 1518, in _perform
    exit_code, output = self._handle_program_exit(child_pid,
  File "/usr/lib/python3/dist-packages/dialog.py", line 1484, in _handle_program_exit
    self._wait_for_program_termination(child_pid,
  File "/usr/lib/python3/dist-packages/dialog.py", line 1430, in _wait_for_program_termination
    raise DialogError(
dialog.DialogError: dialog-like terminated due to an error: the dialog-like program exited with status 3 (which was passed to it as the DIALOG_ERROR environment variable). Sometimes, the reason is simply that dialog was given a height or width parameter that is too big for the terminal in use. Its output, with leading and trailing whitespace stripped, was:

```
This error is apparently caused by [line 129](https://github.com/peterkvt80/vbit2/blob/6ec1769bc0a229059320ce7e6d0da6688136a97b/scripts/vbit-config.py#L129C3-L129C3) of vbit2/scripts/vbit-config.py where the parameter `height=7` is too short for the directory dialog. Changing this to `height=18` allows the directory dialog to appear, allowing installation of the custom service to complete correctly.

ZXGuesser, 2023-08-27:
Hmm, this is very strange. On Buster that "height" is controlling the number of lines of output inside the directory select box, and so something like 18 crashes as the overall box is then vastly too big for a default 80x24 terminal.

gkthemac, 2023-08-27:
Still got my previous Buster install so I had a look, and there definitely seems to be a difference between how Buster and Bullseye handle the "height" parameter.

This is Buster running in a terminal that's 29 lines high, with the height altered to `height=18`. The height parameter seems to control how many entries appear in the "list of directories" pane, so there's 18 lines from the "." to the "Pictures".
[image]

This is Bullseye, again with the height altered to `height=18`. This time the height parameter seems to control the overall height of the entire box, one can count 18 lines from the "Enter teletext service directory" at the top of the box to the black line outline below the OK and Cancel buttons.
[image]

ZXGuesser, 2023-08-27:
Out of interest, what happens if you omit height parameter on Bullseye?

gkthemac, 2023-08-27:
On Bullseye if the "height" parameter is omitted then it works properly, the whole box seems to be 17 lines high regardless of the terminal size. The same happens on height=0, and with height=-1 the dialog is at tall as possible.

On an 80x24 terminal Buster seems to tolerate a maximum of height=14, any higher will crash it. On Bullseye the minimum seems to be height=12 which gives just two lines in the directory list pane!

If it helps, cdialog on Buster is version 1.3-20190211 and on Bullseye it's version 1.3-20201126. One could use the [DialogBackendVersion](https://pythondialog.sourceforge.io/doc/DialogBackendVersion.html) call to find out which it is.

ZXGuesser, 2023-08-27:
Does the workaround in the above branch work?

gkthemac, 2023-08-27:
Yes, works fine on both the Buster and the Bullseye installs.

closed 2023-08-29 by gkthemac