peterkvt80/vbit2 #38: windows tcp bug?

tenkavideo, 2023-11-12:
Hello, I'm encountering an issue with Ncat on Windows. Whenever I send a "P100" command or any other page command, the vbit2 application closes the connection, and it becomes impossible to reconnect until the process is restarted
tries on windows 11,telnet,ncat ,putty
thx!

SonnyWalkman, 2023-11-17:
Hello @tenkavideo and @peterkvt80 

I believe the issue lies in the use of this function `DieWithError` in the `TCPClient` class immediately terminates the process upon encountering an error, as indicated by the `exit(1)` call. 

I believe this is the reason why Vbit2 is needing a restart after an error occurs cause it's dead and unable to listen and establish a new TCP connection.

A more graceful approach might be to handle the error **without** exiting the entire process, perhaps by issuing a command error continuing to be listening on the current client connection.  The Handler function is responsible for managing incoming client connections. It reads data from the client socket and processes it. If an error occurs during data reception (as indicated by `recv() `returning a negative value), signals TCP disconnect the offending `DieWithError` function is called, leading to the termination of the Vbit2 process.

Peter, could, you possible look my change to the  TCPClient function to handle the error? replace the `DieWithError` function to avoid terminating the entire process. Instead, process a disconnect? This approach will enable Vbit2 to handle re-establishing TCP connection without needing a restart. The command handler loop in the Handler function may need extra checks against client disconnections and errors. Add additional error handling to manage different types of exceptions or unexpected behaviours from the client side which I haven't looked into? May need to handle malformed requests?

Replace below:-

```
/** HandleTCPClient
 *  Commands come in here.
 * They need to be accumulated and when we have a complete line, send it to a command interpreter.
 */
void TCPClient::Handler(int clntSocket)
{
    char echoBuffer[RCVBUFSIZE];        /* Buffer for echo string */
    char response[RCVBUFSIZE];
    int recvMsgSize;                    /* Size of received message */
    int i;
    clearCmd();

    /* Send received string and receive again until end of transmission */
    for (recvMsgSize=1;recvMsgSize > 0;)      /* zero indicates end of transmission */
    {
        /* See if there is more data to receive */
        if ((recvMsgSize = recv(clntSocket, echoBuffer, RCVBUFSIZE, 0)) < 0)
            **DieWithError("recv() failed");**
        for (i=0;i<recvMsgSize;i++)
        {
            addChar(echoBuffer[i], response);
            if (*response)
                send(clntSocket, response, strlen(response), 0);
        }
        /* Echo message back to client */
        //if (send(clntSocket, echoBuffer, recvMsgSize, 0) != recvMsgSize)
        //    DieWithError("send() failed");
    }

#ifdef WIN32
    closesocket(clntSocket);
#else
    close(clntSocket);    /* Close client socket */
#endif // WIN32
}
```
With:

```

/** HandleTCPClient
 *  Commands come in here.
 * They need to be accumulated and when we have a complete line, send it to a command interpreter. 
 * If the connection receives a negation or zero value with just close the socket, keep the client running. #bugfix
 */
void TCPClient::Handler(int clntSocket)
{
    char echoBuffer[RCVBUFSIZE];
    char response[RCVBUFSIZE];
    int recvMsgSize;

    clearCmd();

    while (true) 
    {
        recvMsgSize = recv(clntSocket, echoBuffer, RCVBUFSIZE, 0);

        if (recvMsgSize < 0) {
            // Handle the error without exiting the process
            // Optionally log the error or send a message back to the client
            std::cerr << "Error in receiving data." << std::endl;
            break;
        } 
        else if (recvMsgSize == 0) {
            // Client has closed the connection
            std::cerr << "Client has closed the connection." << std::endl;
            break;
        }

        // Process the received data
        for (int i = 0; i < recvMsgSize; i++) {
            addChar(echoBuffer[i], response);
            if (*response) {
                send(clntSocket, response, strlen(response), 0);
            }
        }
    }

#ifdef WIN32
    closesocket(clntSocket);
#else
    close(clntSocket);
#endif
}
```

tenkavideo, 2023-11-17:
thx so much!
the issue in windows is that if for exanple if i send  "P100" command,vibit try to look for Page 10000 then crash the the vbit process

SonnyWalkman, 2023-11-17:
Hopefully, @peterkvt80 or @ZXGuesser  could at least patch TCPclient not kill vbit2 process for the disconnect. However, I'd prefer all errors to be handled with a response.

ZXGuesser, 2023-11-17:
This code hasn't been significantly worked on for a very long time and there's been some fairly large changes to the pagelists since then to solve some hard crashes, which may have broken something this code was trying to do.
I would note though that the last commit message for selecting and modifying pages in the command handler reads `Not working correctly.` so maybe it's always been like this

tenkavideo, 2023-11-17:
Hello Z
Do I think will possible to make it working?
R and L command work but the bog issue is that I cannot select a page with
P command

ZXGuesser, 2023-11-17:
I believe the command interface is working fine and the call to pageList::Match is crashing

tenkavideo, 2023-11-17:
I think so, somebody can fix this part?
I need it for send subtitles to a page for give accessibility to a Church
channel

SonnyWalkman, 2023-11-17:
Hello again @tenkavidee. 

Until vbit2 remote 'P command' is working? Have you tried ffmpeg SRT (SubRip) and mux it directly from a file? https://www.bannerbear.com/blog/how-to-add-subtitles-to-a-video-file-using-ffmpeg/

Just a thought? Or are you injecting on the fly?

tenkavideo, 2023-11-17:
Hello S
I need to do on live events so I need to do it on the fly,so teletext will
be perfect for that

SonnyWalkman, 2023-11-18:
Hello @t,
Ok, yes direct injection as teletext is far easier for your needs.

Here in Australia subtitles are mandatory and injected as teletext on page
801.  I know this by analysing the DVB transport stream in detail on the
private PES.

If both the TCPClient  handling to stop killing the process and the what
ZXGuesser has found with  page handler I believe vbit2 will do serve most
people’s needs.

Peter could look into sorting the page issues and refactoring the TCPClient
function?

I could do it myself however I haven’t set up a dev for C++

 I very happy ZXGuesser as added the TS encapsulation!

A BIG Thankyou from me ZXGuesser!!

Sonny

tenkavideo, 2023-11-18:
Yes very happy for zx work!
i hope can found solution for the tcp cmd issue

SonnyWalkman, 2023-11-24:
> I believe the command interface is working fine and the call to pageList::Match is crashing

`int PageList::Match(char* pageNumber)
{
    int matchCount=0;

    std::cerr << "[PageList::FindPage] Selecting " << pageNumber << std::endl;
    int begin=0;
    int end=7;

    for (int mag=begin;mag<end+1;mag++)
    {
        // For each page
        for (std::list<TTXPageStream>::iterator p=_pageList[mag].begin();p!=_pageList[mag].end();++p)
        {
            TTXPageStream* ptr;
            char s[6];
            char* ps=s;
            bool match=true;
            for (ptr=&(*p);ptr!=nullptr;ptr=(TTXPageStream*)ptr->Getm_SubPage()) // For all the subpages in a carousel
            {
                // Convert the page number into a string so we can compare it
                std::stringstream ss;
                ss << std::hex << std::uppercase << std::setw(5) << ptr->GetPageNumber();
                strcpy(ps,ss.str().c_str());
                // std::cerr << "[PageList::FindPage] matching " << ps << std::endl;

                for (int i=0;i<5;i++)
                {
                    if (pageNumber[i]!='*') // wildcard
                    {
                        if (pageNumber[i]!=ps[i])
                        {
                            match=false;
                        }
                    }
                }
            }
            if (match)
            {
                matchCount++;
            }
            ptr->SetSelected(match);
        }
    }
    // Set up the iterator for commands that use pages selected by the Page Identity
    _iterMag=0;
    _iter=_pageList[_iterMag].begin();
    
    return matchCount; // final count
}`

**pageNumber** passed in most likely needs to be checked for range? 100-899?  Return error is out of range.  @peterkvt80 maybe able to look at why the function crashes?

tenkavideo, 2023-11-24:
Hello Rob
Thx for take care of this
On depends of number of pages also L and R crash for the same reason
Hope can be fix!

ZXGuesser, 2023-12-09:
0b8613db2d1ec2387c81e41dc40203398002f503 fixes the crash in FindPage::Match caused by it accessing a null pointer rather than the subpages it was (presumably) intending to.

There's still a lot of dragons lurking in this code that will crash the process, caused by missing bounds checking and the like.

SonnyWalkman, 2023-12-13:
Thanks for your work @ZXGuesser,
Is there any chance in sorting out Vbit2 terminating on error or client disconnection?

ZXGuesser, 2023-12-13:
I've been unable to replicate a broken socket and whenever I disconnect it goes back to `Ready for a client to connect` as expected. It would help to know what you actually do to cause the error.

tenkavideo, 2023-12-13:
Hello!

If u can provide the windows binary I can test for u