peterkvt80/vbit2 #40: getvbit2 for Orange Pi (Zero 2) or similar possible?

Paderman, 2024-04-17:
Hello,
is it possible to make a port for Orange Pi (Zero 2) or similar?
Orange Pi Zero 2 has HDMI and 3.5mm video/audio output.
Thank you.

avrovulcanxh607, 2024-04-17:
Hi, 

I've played around with this on the Zero LTS - unless things have changed, you need a weird firmware build to even get composite video to work - and since the system is fundamentally different to a Raspberry Pi, it would need a bespoke program to match, unfortunately.

So, possible? - maybe
Likely to happen? - not really 😅

I used a custom hardware board to do teletext on my Orange Pis, but to be honest it's not worth the effort except in very specific circumstances - I'd just use a normal raspberry pi!

Paderman, 2024-04-20:
Hi,
OK, thanks for the info :)

What _Raspberry Pi's_ are compatible?

What for a custom board do you use :D

avrovulcanxh607, 2024-04-21:
Any Raspberry Pi should work - just install the "legacy" OS, use the composite video output (you'll probably need an annoying 3.5mm to phono cable), and install vbit2 as described in the wiki :)

Paderman, 2024-04-21:
OK, I will try if I enough have money to buy a _Raspberry Pi_ 😉

The _Orange Pi Zero 2_ has with an additional module (Orange Pi Expansion Board) a/v output, 4 PIN, Video + Audio (2 stereo) 😇 🤷