peterkvt80/vbit2 #44: Error mapping register memory

cwgh1, 2025-04-10:
I am humbled by what this project seems to be able to do, and I decided to start learning about Teletext with the aim of creating a set of pages which are updated periodically from some environmental data I collect.

To get going I purchased a new Raspberry Pi Model 4B with a 64Gb MicroSD card, a suitable cable and an old CRT TV which has a composite video input and supports teletext. But I have been struggling to get it working consistently. 

My first attempt was installing "Raspberry Pi OS (Legacy, 64 bit) Full (Bullseye)", and then following the instructions at https://github.com/peterkvt80/vbit2/wiki 

```
sudo apt-get update
sudo apt-get upgrade
```

```
cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"...
```

```
sudo nano /boot/config.txt
```

Comment out "dtoverlay=vc4-kms-v3d" -> "#dtoverlay=vc4-kms-v3d"

Save changes.

```
sudo reboot now
```

Connect to Pi by SSH (as above)

Run config

```
sudo raspi-config
```

Set Display Options > D6 Composite > V2 enable composite
and Display Options > Underscan > No

Exit and confirm reboot.

```
curl https://raw.githubusercontent.com/peterkvt80/vbit2/master/getvbit2 | bash
```

After running the configurator and selecting a service like Ceefax - London, or Teefax, success!

I am able to see pages displayed on the screen, but it is somewhat garbled as if different parts are getting overwritten and other parts not:

[image]

I wondered if the fragmented pages might be a configuration thing, but I haven't looked into that too much yet, because **after any subsequent reboot**, I have never been able to see any teletext output - it is like there is no service/signal:

[image]

The configurator still executes, so I tried changing settings in the VBIT configurator, config files, and the Pi configurator, but never got back to see these fragmented pages. You can select to start and stop the service in the configurator, but not sure it is doing that.

I gave up for a few weeks. To try to rule out SD card issue or OS, I decided to do a fresh OS and code install on a new 64Gb microSD card, this time with the Lite Legacy 32bit version (Bullseye).

Again, that worked (but with fragmented pages appearing), **until reboot** and then again nothing. So no different to the 64bit full OS. Looking in /var/log/syslog I see some error mentioned when starting or stopping VBIT2:

```
Apr 10 17:26:40 pioraclite systemd[486]: Started VBIT2.
Apr 10 17:26:40 pioraclite runvbit2[4339]: Error mapping register memory.
Apr 10 17:26:46 pioraclite systemd[486]: Stopping VBIT2...
Apr 10 17:26:46 pioraclite runvbit2[4352]: Error mapping register memory.
Apr 10 17:26:46 pioraclite systemd[486]: vbit2.service: Succeeded.
Apr 10 17:26:46 pioraclite systemd[486]: Stopped VBIT2.
```

I then tried to run from command line, but that also responds with an error:

```
python vbit2/scripts/runvbit2.py
Error mapping register memory.
^CError mapping register memory.
```

I wonder if that error message matters and if that is contributing to my inability to see teletext pages following the first shutdown/reboot after successfully seeing fragmented pages. 

Does "Error mapping register memory" matter? I am not a C++ programmer nor that very confident with Linux, but have programmed in other languages (Basic, Fortran, Perl, JavaScript, Python and a bit of PHP). If the error matters, is there anything else I can do to investigate what is causing it? I don't have a spare Pi or another teletext device, but could make a purchase to eliminate hardware matters? Or do I need an older OS version perhaps?

Any thoughts from those more experienced would be warmly received to help me get going.

ZXGuesser, 2025-04-10:
I believe the `error mapping register memory` is from the raspi-teletext tvctl program which is what modifies the video output hardware registers to put teletext signals into the vertical blanking interval.
I know people have had issues with this on 64 bit OS, but 32 bit should work fine in my experience...

I notice from your photo that the teletext display is disappearing off the bottom of the screen which puzzles me. Can you confirm that the pi is generating a 625 line video signal? (`sdtv_mode=2` in the boot config)

cwgh1, 2025-04-11:
Thank you @ZXGuesser very much.

In boot config, yes this mode is already set as '2':

> sdtv_mode=2

I started again (same Pi, same SDcard, same TV) and installed the OS and VBIT2 from scratch. I double-checked I was installing "Raspberry Pi OS (Legacy, 32-bit) Lite" selected from the Raspberry Pi Imager application (see image below).

This time I examined whether any errors are reported while I can see a working teletext service (the fragmented pages), before I shutdown/reboot. The "Error mapping register memory" message does appear even at this stage when the pages are visible (before subsequent reboot when the pages are no longer visible).

> Apr 11 11:00:52 pioraclite2 dhcpcd[629]: wlan0: Router Advertisement from fe80::ba6a:f1ff:fe83:a6e0
> Apr 11 11:01:12 pioraclite2 systemd[665]: Reloading.
> Apr 11 11:02:42 pioraclite2 systemd[665]: Started VBIT2.
> Apr 11 11:02:43 pioraclite2 runvbit2[2226]: Error mapping register memory.

In my original previous message, I did not know if the error message had occurred only after reboot; no, it is also before, so perhaps not the main problem here. After the next reboot, VBIT2 is running (option set to auto start), but after selecting TV/AV, then pressing the Text button, no pages displayed in Teletext. This is consistent with my previous two OS install attempts.

[image]

My next steps: I am going to get a Model 3+ Pi and try that instead of my Model 4. And am also trying to source a different CRT teletext TV, in case the page format problem is related to the Panasonic TV. I will report back in about a week...

ZXGuesser, 2025-04-11:
The teletext signal should not be visible at the top of the screen, this is because the tvctl program is failing to modify the graphics output register to shift the video upwards.
It seems the TV is reading teletext from the first of the picture lines.

cwgh1, 2025-04-16:
Findings after I obtained another teletext TV and a new Raspberry Pi 3 Model B+:

- Both teletext TVs perform the same problematic way with the original Raspberry Pi 4 Model B using either legacy 64bit full or legacy 32bit Lite;
- Using the Pi 3 Model B+ (using the same installation routine and configuration settings) with legacy 32bit Lite **works correctly**, and continues to work after reboot.

So, in summary, problem solved by using a 3 Model B+ instead of 4 Model B.

Thank you very much for your input, which encouraged me to pursue the problem further. I will close this issue and now progress with what I want to do... developing some pages and getting a service on display somewhere.

It's fantastic to have such clever software available, thank you.

closed 2025-04-16 by cwgh1

avrovulcanxh607, 2025-04-16:
just to add, given that recent Pi OS break raspi-teletext - and I'm not aware of any plans to patch it - it may be a good idea to post a link to a "known good" OS in RPi's legacy downloads portal thingy, eg I use https://downloads.raspberrypi.org/raspios_oldstable_lite_armhf/images/raspios_oldstable_lite_armhf-2022-09-26/
for OMXPlayer and IIRC it works well for teletext on all Pis too.