peterkvt80/vbit2 #48: Stream capture

odehalaa, 2025-06-01:
Hi All,

please I need to capture teletetxt stream this way correct or not please need to inject mux to dvb stream  ?

./vbit2  --dir /home/pi/teletext --pid 154 --format ts  > output.ts

Regards

ZXGuesser, 2025-06-01:
Using vbit2 for DVB teletext is quite difficult to set up, and the following assumes you're already an expert on DVB.

I've had moderate success using [tsduck](https://tsduck.io/) to inject the teletext transport stream using the tsp `merge` plugin with the `--incremental-pcr-restamp` option.
You will then need to patch the PMT with the `pmt` plugin to associate the pid of the teletext stream with the program.

odehalaa, 2025-06-01:
Thanks a lot for your kindly support, yes please to get stream output or any format to inject it by tsduck ?

Regards

odehalaa, 2025-06-01:
from vbit how we can get live stream or any output format or need hardware to inject teletext to tsduck ?

ZXGuesser, 2025-06-01:
It's as you had, `--pid 154 --format ts` will generate a transport stream with pid 154 on stdout and the tsp merge plugin will consume it. Your tsp command would include something like `-P merge "/home/pi/vbit2/vbit2 --dir /home/pi/teletext --pid 154 --format ts" --accel 0 --incremental-pcr-restamp`
I'm afraid you'll have to study the tsduck documentation to figure out how to make this applicable to your use case, and how to patch the PMT etc as it's far too complicated to try and describe how it all works here and will depend on where your existing transport stream is coming from, and where you are trying to send it etc.

odehalaa, 2025-06-05:
Sorry ZXGuesser, is there full step and procedure please to get DVB teletext  ?

odehalaa, 2025-06-05:
DVB-TXT is there any example please ?

odehalaa, 2025-06-05:
Thanks a lot ZXGuesser for your kindly support

odehalaa, 2025-06-06:
Hi, 

I used this command and format ts and tsnpts with pid 256 and without pid  aslo
/home/pi/.local/bin/vbit2 --dir /home/pi/.teletext-services/CeefaxLondon  --pid 256  --format ts > /home/pi/output.ts

but still the ts not contain the subtitle please how we can get it ?

Input: /home/pi/output.ts
[Extract: 1] [Stream mode: Autodetect]
[Program : Auto ] [Hauppage mode: No] [Use MythTV code: Auto]
[Timing mode: Auto] [Debug: No] [Buffer input: No]
[Use pic_order_cnt_lsb for H.264: No] [Print CC decoder traces: No]
[Target format: .srt] [Encoding: UTF-8] [Delay: 0] [Trim lines: No]
[Add font color data: Yes] [Add font typesetting: Yes]
[Convert case: No] [Video-edit join: No]
[Extraction start time: not set (from start)]
[Extraction end time: not set (to end)]
[Live stream: No] [Clock frequency: 90000]
[Teletext page: Autodetect]
[Start credits text: None]
[Quantisation-mode: CCExtractor's internal function]

-----------------------------------------------------------------
Opening file: /home/pi/output.ts
File seems to be a transport stream, enabling TS mode
Analyzing data in general mode
Premature end of file - Transport Stream packet is incomplete (expected 188 bytes, got 124).
Done, processing time = 0 seconds

No captions were found in input.
Issues? Open a ticket here
https://github.com/CCExtractor/ccextractor/issues

Regards