peterkvt80/vbit2 #49: vbit.config not woking on custom services.

midelgadinho, 2025-07-11:
I've installed a fresh copy of vbit2 on my Raspberry Pi and I've been facing the following issue: The header shows as the vbit2 default and the status_display is not showing. This is only happening to custom services as the "Ceefax" and "Teefax", for example, are working correctly.
I've tried with different teletext "custom services" and differente configs, but the result stays the same.

I've referred to this issue (https://github.com/peterkvt80/vbit2/issues/28) to try and solve this problem but it yielded no result. Also, worth noting that the OS is a fresh install (Raspbian Bullseye, last version).

Thank you for the time and the attention!

avrovulcanxh607, 2025-07-11:
Could you share your vbit.conf & explain your expected results please

ZXGuesser, 2025-07-11:
Are you confusing status_display and header_template? Many TVs never display the status_display text anywhere.

midelgadinho, 2025-07-11:
I'm currently using a "modified" example vbit.conf supplied here on this Github. It used to work until last year (the last time I've used this software) and, yeah, my TV still shows, for example, the CEEFAX status_display. 

Here follows my vbit.conf:

`; example vbit config file
; lines beginning with a semicolon are comment lines
; settings are in the form name=value and there must be no whitespace.

;--------------------------------- HEADER ROW ---------------------------------
; header template value must be 32 teletext characters long.
; both 8-bit control codes and 7-bit tti escape sequences are supported.
; %%# is replaced by the page number
; time and date use a subset of posix strftime conversion specifier characters
; pre-padded with % symbols to the appropriate length.
; day name = %%a
; month name = %%b
; day number with leading zero = %d
; day number without leading zero = %e
; month number = %m
; 2 digit year = %y
; hour = %H
; minute = %M
; second = %S
; here are some historic examples of header rows:
;header_template=CEEFAX %%#  %%a %e %%b C%H:%M/%S
;header_template=CEEFAX 1 %%# %%a %d %%bC%H:%M/%S
;header_template=ORACLE %%# %%a%e %%b ITVB%H%M:%S
;header_template=ORACLE %%# %%a%e %%bCITV %H%M:%S
;header_template=Teletext on 4 %%# %%b%dC%H:%M:%S
;header_template=D]CTeletext G\%%# %%b%dC%H:%M:%S

; RTVG header
header_template= BRTVGtexto G%%a %e%%b %H:%M:%S

;---------------------------- INITIAL TELETEXT PAGE ---------------------------
; magazine, page, and subcode to transmit within broadcast data service packets
; magazine and page number are specified as you would on a teletext set.
; i.e. magazine 1-8 followed by two hex digits for example 100, 888, 19F, etc.
; the initial subcode can optionally be appended, separated by a colon.
initial_teletext_page=100
;initial_teletext_page=100:0000

;------------------------------ SERVICE SETTINGS ------------------------------
; omit blank rows to increase transmission efficiency (defaults to false)
;row_adaptive_mode=false

; specify number of VBI lines per video field
;lines_per_field=16

; set the priority of each magazine. 1=highest priority, 9=lowest.
; eight comma separated values for magazines 8,1,2,3,4,5,6,7.
;magazine_priority=9,3,3,6,3,3,5,6

; 20 character status message for broadcast service data packet
status_display=RTVG Portugal

; 16 bit hexadecimal code assigned to network in ETSI TR 101 231
; defaults to 0000 (no NI code assigned)
;network_identification_code=0000

; 16 bit hexadecimal code assigned for PDC
;country_network_identification=0000

;------------------------------ SUBTITLE SETTINGS ------------------------------
; number of times that a subtitle should be retransmitted. 0..9, default 1
;subtitle_repeats=1

;--------------------------- COMMAND INTERFACE SETTINGS ------------------------
; TCPIP control interface (defaults to disabled)
;enable_command_port=false
;command_port=5570`