peterkvt80/vbit2 #51: No longer builds on Raspbian

raboof, 2025-09-16:
v2.7.3 was still fine, but for v2.8.1:

```
/usr/bin/ld: interfaceServer.o: in function `vbit::InterfaceServer::InterfaceServer(vbit::Configure*, vbit::Debug*, vbit::PageList*)':
/home/pi/vbit2/interfaceServer.cpp:41: undefined reference to `vbit::PacketDatacast::PacketDatacast(unsigned char, vbit::Configure*)'
/usr/bin/ld: interfaceServer.o: in function `vbit::InterfaceServer::run()':
/home/pi/vbit2/interfaceServer.cpp:338: undefined reference to `vbit::PacketDatacast::PushIDLA(unsigned char, unsigned char, unsigned int, unsigned char, unsigned char, std::vector<unsigned char, std::allocator<unsigned char> >*)'
/usr/bin/ld: /home/pi/vbit2/interfaceServer.cpp:306: undefined reference to `vbit::PacketDatacast::PushRaw(std::vector<unsigned char, std::allocator<unsigned char> >*)'
collect2: error: ld returned 1 exit status
make: *** [Makefile:23: vbit2] Error 1
```

I'm not sure I'll get to bisecting, it's rather painful on this slow Raspberry Pi 1 :)

ZXGuesser, 2025-09-16:
That's peculiar, it's a Pi 1 that I built it on here for testing. 🤔

ZXGuesser, 2025-09-16:
Mine will have something ancient on it because the newer versions are a pain to do video things. I've just kicked off a build on one of the zeros I need to update but I have them underclocked so it's an even slower process!

ZXGuesser, 2025-09-16:
Have you tried building from clean or only a rebuild in place?

raboof, 2025-09-17:
This was a new intallation (after a hardware failure with the previous https://hack42.nl/wiki/Gebruiker:Raboof/TeleMatrikst), so 'fairly clean' - I think there may have been a previous aborted build due to the power supply being fiddly, but this machine has never seen a different version of vbit2

ZXGuesser, 2025-09-17:
After a lot of pissing about getting logged in, I've just done an install on a brand new 32 bit raspios bookworm lite image on my Pi 1 with no errors so I don't know what to suggest sorry.

raboof, 2025-09-17:
Ugh, sorry to send you on this wild goose chase. I used a non-lite image but that seems unlikely to be the difference. I'll close this for now and will try to further diagnose the issue when the pi is back up and running.

Thanks a lot for an awesome tool, we're having a lot of fun showing our Matrix chat channel logs and event calendar on Teletext 😄

closed 2025-09-17 by raboof